mr-canoehead/vpn_client_gateway #7: Disabling internet connectivity when VPN connection

fallenidol, 2015-09-05:
The guide makes no mention of what happens when the VPN connection goes down or is not active. Ideally I would like internet connectivity to drop out when not connected via the VPN so that no unsecured traffic goes out over the ISP connection. Is this something you could consider adding.

mr-canoehead, 2015-09-06:
The solution already handles this; the NAT rule in the firewall script forwards all traffic through the VPN network adapter (tun0). If the VPN session drops, packets will fail to be forwarded and any sessions will break. However, there is still potential for DNS queries to be exposed, since dnsmasq running on the Pi may forward queries via the ethernet adapter (eth0) if the VPN drops. A workaround for this is to use Google's DNS (8.8.8.8, 8.8.4.4) servers on client machines as the DNS queries will then be forwarded via VPN like all other traffic (or not forwarded, if the VPN connection drops). I should look in to the DNS forwarding issue, there may be a way to manage this via dnsmasq configuration.

*Note: the user does have the option of manually turning off VPN which will expose all forwarded traffic. This is done via the Admin tab or via a special VPN server entry  (refer to [Disabling VPN](https://github.com/mr-canoehead/vpn_client_gateway/wiki/Disabling-VPN) in the wiki).
When the user disables VPN, the NAT rule is changed to forward all traffic via the ethernet adapter (eth0). When the user enables VPN, the NAT rule is changed back to forwarding via VPN (tun0).

I don't have time to look in to the DNS forwarding right now, so I'll leave this issue open until I get to it in a few weeks. I should also update the wiki so that it's clear what the behaviour is.

mr-canoehead, 2017-02-28:
Better late than never; I have updated the firewall rules to include a 'kill switch' feature which blocks network traffic in the event that the VPN connection drops. This includes network traffic originating from the VPN Client Gateway (i.e. not just forwarded traffic). I've added a page describing this behavior: [Kill Switch feature](https://github.com/mr-canoehead/vpn_client_gateway/wiki/Kill-Switch-feature).

Thanks for your suggestion, hopefully this addresses your concerns with the project; any feedback is appreciated.

Cheers,
Chris

closed 2017-02-28 by mr-canoehead