mr-canoehead/vpn_client_gateway #10: SOCKS5 Proxy Server

currowth, 2015-11-01:
Can you add support for socks5 proxy server? It would be great for getting specific applications to go through the vpn.

mr-canoehead, 2015-11-03:
Can you please describe the setup you envision? Haven't worked with socks5 at all, so I'm not familiar with it. Would it be socks5 connections instead of VPN (i.e. switching servers connects to a different socks5 server), or a socks5 proxy running on the Pi accepting connections and forwarding traffic to a single socks5 server over the VPN (same socks5 server regardless of which VPN server is used)?

Thanks,
Chris

mr-canoehead, 2015-11-11:
Closing issue due to lack of feedback. Feel free to email me if you wish to discuss this feature request.

closed 2015-11-11 by mr-canoehead