mr-canoehead/vpn_client_gateway #13: Apple TV

ClaudioBocca, 2016-05-13:
Hi. I built the Pi Router  with the web interface and a PureVPN service. Apparently the router is working fine but NETFLIX on the Apple TV recognise that a VPN service is running so it locks any media streaming. Anyway the project is really well explained and it works really well. This info is to let You update the section regarding the Apple TV.
Thank You.
Claudio

mr-canoehead, 2016-05-13:
Hi Claudio,
Thanks for the feedback, I have updated the media streamer notes for Apple TV and Chromecast to include a note that VPN access to some providers (Netflix) is now blocked.

Cheers,
Chris

closed 2016-05-13 by mr-canoehead

ClaudioBocca, 2016-05-13:
Unfortunately... I should add…

Cheers,
Claudio