mr-canoehead/vpn_client_gateway #21: Enabling UPNP?

rmclellan, 2017-03-24:
Working on UPNP for port forwarding.  Tried the following with no luck:

1
sudo apt-get install linux-igd

sudo nano /etc/openvpn/tunnel.up

```
#!/bin/sh
/usr/sbin/upnpd tun0 eth0
```

sudo nano /etc/openvpn/tunnel.down

```
#!/bin/sh
/usr/bin/killall upnpd
```
Make the scripts executable:

sudo chmod +x /etc/openvpn/tunnel.up /etc/openvpn/tunnel.down

server.conf
```
# add up and down script for uPNP
script-security 2
up /etc/openvpn/tunnel.up
down /etc/openvpn/tunnel.down
```

```
sudo iptables -A INPUT -i eth0 -p udp -m multiport --dports 1900,5351,5353 -j ACCEPT
sudo iptables -A INPUT -i eth0 -p tcp -m multiport --dports 49152 -j ACCEPT
sudo netfilter-persistent save
```
[Hackviking](https://www.hackviking.com/single-board-computers/pi-make-a-vpn-gateway-with-upnp-port-forwarding/)

mr-canoehead, 2018-05-15:
I have recently done some work on port forwarding that may be relevant to this issue, you can check out the new port forwarding wiki page here:
[https://github.com/mr-canoehead/vpn_client_gateway/wiki/Port-forwarding](https://github.com/mr-canoehead/vpn_client_gateway/wiki/Port-forwarding)

Can you please provide an example use case of how UPnP would be used with the VPN Client Gateway?

Thanks,
Chris

mr-canoehead, 2019-04-07:
Closing stale issue, please open a new issue if you have new info to add.

closed 2019-04-07 by mr-canoehead