mr-canoehead/vpn_client_gateway #28: Protonvpn

ismay, 2017-12-25:
I’d like to try and add support for protonvpn. I’ll be running the gateway on an odroid c2, but the basics should be the same.

Has anyone already tried adding protonvpn? Any tips or things I should look out for?

mr-canoehead, 2017-12-29:
I haven't heard of anyone using ProtonVPN yet. Assuming they use OpenVPN it should be straightforward to add support - just create the appropriate server.conf and vpnservers.xml files. One catch could be if ProtonVPN requires a different certificate for each of their servers; in this case, code may need to be changed so that the openvpn service uses the correct cert depending on which server is chosen. Robert McLellan made some changes to the manage_openvpn.php script to support NordVPN which requires a unique cert for each server, so it would be useful to look at that code if ProtonVPN also requires unique certs.

ismay, 2017-12-30:
So I worked on it a bit, and they do require unique certificates for each server (they’re embedded in the the .ovpn). Also, they’re pushing routes that are messing with the routing of client traffic (a ‘curl icanhazip.com’ from the gateway works as expected when the vpn is up, but client traffic isn’t being routed), so I’ve submitted a support request and am waiting for their response.

I’ll update if there’s new info and will submit a PR if I have something that works.

ismay, 2018-01-20:
So protonvpn hasn't responded yet, and frankly I don't want to devote too much time to it, so I'm just gonna go with a proprietary solution. Probably a router with openvpn client support built in. I'm closing this because I'm not going to work on it anymore, but feel free to reopen if anyone else wants to!

closed 2018-01-20 by ismay