mr-canoehead/vpn_client_gateway #29: Iptables rules and Vpn Kill Switch for Wlan0

farisoftfry, 2018-01-16:
mr-canoehead your project is wonderful. And it works like a charm. But, for many reason, I need to use the built-in wifi of Raspberry Pi 3 to share my Vpn Internet connection (through wifi, of course) to other devices.
I used thas project to install the access point and it works very well: https://github.com/billz/raspap-webgui. The only problem is that  webgui has the same IP address so I can see just the last webgui installed.
But the big problem for me (I'm not a linux expert) is to configure iptables to forward connection from tun0 to wlan0 and create right Vpn Kill Switch rules, like in your project.
I tried to add to your rules these one:
-t nat -A POSTROUTING -o tun0 -j MASQUERADE
-A FORWARD -i tun0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -i wlan0 -o tun0 -j ACCEPT
but, together with your rules, they don't work.
But If I clear your rules through your script and manually I add these rules they work very good.
The problem is that, in this way, there is not any Vpn kill switch and I would like to have a fully functional Vpn (with kill switch) on my Pi to use as a gateway/dns server (as in your project) and a wireless access point. Could you help me? And thanks a lot for sharing your project.

mr-canoehead, 2018-01-17:
I recently acquired a Pi 3, so I should be able to take a look at forwarding WiFi via VPN soon. As to the webgui issue, you should be able to copy the VPN Client Gateway files to a subfolder under the html root folder, e.g. /var/www/html/vpnclgw, then access the management web page via  http://10.1.2.5/vpnclgw (using your Pi's ip address of course).
This is how I have the gateway configured on one of my Pi's. Moving the gateway files to a subfolder should prevent conflicts with other web based stuff that needs to live in the root folder.

farisoftfry, 2018-01-17:
thanks for your reply.
I'm working on the webgui and I will try your advice.
After many attempt I think I've solved the firewall problem in this way (please check if I did something wrong, like if I opened something that I did not have to do, leaks and so on):

`$ sudo nano /etc/iptables/rules.v4`

copy and paste the follow:
`*filter
:INPUT DROP [0:0]
:FORWARD DROP [0:0]
:OUTPUT DROP [0:0]
:BADPKT_LOGGING - [0:0]
:LOGGING - [0:0]
:forward_rules_lan - [0:0]
:forward_rules_vpn - [0:0]
:killswitch - [0:0]
:killswitch_off - [0:0]
:killswitch_on - [0:0]
-A INPUT -i wlan0eth0 -m comment --comment "In from LAN" -j ACCEPT
-A INPUT -i wlan0wlan -m comment --comment "In from LAN" -j ACCEPT
-A INPUT -i lo -m comment --comment loopback -j ACCEPT
-A INPUT -p tcp -m tcp ! --tcp-flags FIN,SYN,RST,ACK SYN -m state --state NEW -j BADPKT_LOGGING
-A INPUT -m state --state INVALID -j BADPKT_LOGGING
-A INPUT -p tcp -m tcp --tcp-flags FIN,SYN,RST,PSH,ACK,URG NONE -j BADPKT_LOGGING
-A INPUT -p tcp -m tcp --tcp-flags FIN,SYN,RST,PSH,ACK,URG FIN,SYN,RST,PSH,ACK,URG -j BADPKT_LOGGING
-A INPUT -f -m comment --comment "Drop FRAGS" -j BADPKT_LOGGING
-A INPUT -p tcp -m tcp --tcp-flags FIN,SYN,RST,PSH,ACK,URG FIN,SYN,RST,ACK -j BADPKT_LOGGING
-A INPUT -p tcp -m tcp --tcp-flags FIN,SYN FIN,SYN -j BADPKT_LOGGING
-A INPUT -p tcp -m tcp --tcp-flags SYN,RST SYN,RST -j BADPKT_LOGGING
-A INPUT -i tun+ -m state --state RELATED,ESTABLISHED -j ACCEPT
-A INPUT -m comment --comment "LOG and DROP" -j LOGGING
-A FORWARD -j forward_rules_vpn
-A OUTPUT -o tun+ -m comment --comment "Out to VPN" -j ACCEPT
-A OUTPUT -o lo -m comment --comment loopback -j ACCEPT
-A OUTPUT -o wlan0eth0 -p icmp -m comment --comment icmp -j ACCEPT
-A OUTPUT -o wlan0wlan0 -p icmp -m comment --comment icmp -j ACCEPT
-A OUTPUT -o wlan0eth0 -p udp -m multiport --dports 53,1194,1197,1198 -m comment --comment openvpn -j ACCEPT
-A OUTPUT -o wlan0wlan0 -p udp -m multiport --dports 53,1194,1197,1198 -m comment --comment openvpn -j ACCEPT
-A OUTPUT -o wlan0eth0 -p tcp -m tcp --sport 22 -m comment --comment ssh -j ACCEPT
-A OUTPUT -o wlan0wlan0 -p tcp -m tcp --sport 22 -m comment --comment ssh -j ACCEPT
-A OUTPUT -o wlan0eth0 -p udp -m udp --dport 123 -m comment --comment ntp -j ACCEPT
-A OUTPUT -o wlan0wlan0 -p udp -m udp --dport 123 -m comment --comment ntp -j ACCEPT
-A OUTPUT -o wlan0eth0 -p udp -m udp --dport 53 -m comment --comment dns -j ACCEPT
-A OUTPUT -o wlan0wlan0 -p udp -m udp --dport 53 -m comment --comment dns -j ACCEPT
-A OUTPUT -o wlan0eth0 -p tcp -m tcp --dport 53 -m comment --comment dns -j ACCEPT
-A OUTPUT -o wlan0wlan0 -p tcp -m tcp --dport 53 -m comment --comment dns -j ACCEPT
-A OUTPUT -p udp -m udp --dport 67:68 -m comment --comment dhcp -j ACCEPT
-A OUTPUT -o wlan0eth0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A OUTPUT -o wlan0wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A OUTPUT -j killswitch
-A OUTPUT -m comment --comment "LOG and DROP" -j LOGGING
-A BADPKT_LOGGING -m limit --limit 2/sec -j LOG --log-prefix "IPTables- BADPACKETS: "
-A BADPKT_LOGGING -j DROP
-A LOGGING -m limit --limit 2/sec -j LOG --log-prefix "IPTables-Dropped: "
-A LOGGING -j DROP
-A forward_rules_lan -i wlan0eth0 -o wlan0eth0 -m comment --comment "LAN forwarding" -j ACCEPT
-A forward_rules_lan -i wlan0wlan0 -o wlan0wlan0 -m comment --comment "LAN forwarding" -j ACCEPT
-A forward_rules_vpn -i tun+ -o wlan0eth0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A forward_rules_vpn -i tun+ -o wlan0wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A forward_rules_vpn -i wlan0eth0 -o tun+ -m comment --comment "LAN out to VPN" -j ACCEPT
-A forward_rules_vpn -i wlan0wlan0 -o tun+ -m comment --comment "LAN out to VPN" -j ACCEPT
-A killswitch -j killswitch_on
-A killswitch_off -o wlan0eth0 -j ACCEPT
-A killswitch_off -o wlan0wlan0 -j ACCEPT
-A killswitch_on -j RETURN
COMMIT
# Completed on Wed Jan 17 19:24:2527:09 2018
# Generated by iptables-save v1.6.0 on Wed Jan 17 19:24:2527:09 2018
*nat
:PREROUTING ACCEPT [0:0]
:INPUT ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:POSTROUTING ACCEPT [0:0]
-A POSTROUTING -o wlan0eth0 -j MASQUERADE
-A POSTROUTING -o wlan0wlan0 -j MASQUERADE
-A POSTROUTING -o tun+ -j MASQUERADE
COMMIT`

ctrl X - Y 

copy in the terminal:

`$ sudo iptables -t nat -A POSTROUTING -o tun0 -j MASQUERADE
$ sudo iptables -A FORWARD -i tun0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
$ sudo iptables -A FORWARD -i wlan0 -o tun0 -j ACCEPT
$ sudo sh -c "iptables-save > /etc/iptables/rules.v4"`

mr-canoehead, 2018-01-18:
I've added wlan0 to the firewall rule script, seems to be working as intended. When the VPN is enabled, traffic from wlan0 is forwarded via the VPN tunnel. When the VPN is disabled, traffic from wlan0 is forwarded via the LAN (eth0). You can try out the firewall rules by downloading the script via the link below, extracting with "gzip -d fw-script-wlan0-test.gz" and running it on your Pi "sudo ./fw-script-wlan0-test". Before running the script make sure that the VPN is enabled via the management web page so that the iptables rules and VPN Gateway state will be in sync.

Note that the attached script doesn't include any local network rules (i.e. when the VPN is active your Pi won't be able to access your local network beyond the allowed services: openvpn, dns, etc.), so you'll have to add them if needed.

I'll need to do some work to incorporate this change into the firewall configuration script (the script that generates the firewall rules script), that will have to wait until I have some time to work on it. I'll leave this issue open until that work is done.

Let me know if you encounter any issues in testing the attached script.

[fw-script-wlan0-test.gz](https://github.com/mr-canoehead/vpn_client_gateway/files/1644610/fw-script-wlan0-test.gz)

farisoftfry, 2018-01-21:
thank for your work, I used what I wrote in my last post and it works very good for me but I saw that your script is almost the same like mine, I tested it and I have not any problem!
p.s. I solved the webgui interface problem too thanks to your advice, thanks again

mr-canoehead, 2018-01-27:
I have completed the changes to the firewall configuration script, it now supports multiple network interfaces. I have also written up instructions on setting up the Pi as a wireless access point, those instructions can be found on the following page:
[Configuring the Pi as a WiFi Access Point](https://github.com/mr-canoehead/vpn_client_gateway/wiki/Configuring-the-Pi-as-a-WiFi-Access-Point)

closed 2018-01-27 by mr-canoehead