mr-canoehead/vpn_client_gateway #32: Port Forwarding

lehmanjo, 2018-04-30:
Hi,

it would be great if you could add a simple example of how to add port-forwarding through the firewall.

Thanks

J.

lehmanjo, 2018-04-30:
Should the following be enough?

```
iptables -A PREROUTING -t nat -i eth0 -p tcp --dport 80 -j DNAT --to 192.168.1.2:8080
iptables -A FORWARD -p tcp -d 192.168.1.2 --dport 8080 -j ACCEPT
```
from  https://www.systutorials.com/816/port-forwarding-using-iptables/

Modified to use your script's variable for the VPN interface ($VPN_INTERFACE) and $PORT_FORWARD_PORT, $PORT_FORWARD_DST_IP and $PORT_FORWARD_DST_PORT to describe forwarding.
```
iptables -A PREROUTING -t nat -i $VPN_INTERFACE -p tcp --dport $PORT_FORWARD_PORT -j DNAT --to $PORT_FORWARD_DST_IP:$PORT_FORWARD_DST_PORT
iptables -A FORWARD -p tcp -d $PORT_FORWARD_DST_IP --dport $PORT_FORWARD_DST_PORT -j ACCEPT
```

I was thinking of adding it after the section...

```
# add rule chain for forwarding via LAN
```

Is that all it would take?  

J.

lehmanjo, 2018-04-30:
Implemented a function to generate the above but it doesn't seem sufficient.

[https://pastebin.com/rpeCgH5s](https://pastebin.com/rpeCgH5s)

mr-canoehead, 2018-04-30:
Can you please describe from a high level what you are trying to accomplish, e.g. a use case for the setup you envision?

lehmanjo, 2018-05-02:
Thanks.  

My provider PureVPN seems to not offer any protection for the established VPN tunnel.  They have a separate add-on which can be purchased which adds a "NAT Firewall" on their end.   That's good for me since your VPN client gateway provides me with NAT and a Firewall on my end of the VPN tunnel.  Double NAT would be unnecessarily complicated.

mr-canoehead, 2018-05-04:
I've got a test setup working with port forwarding via my provider (Private Internet Access). Also using a dynamic DNS provider so that I can access via hostname rather than IP address.
The Pi is forwarding incoming TCP connections from the VPN (on the port assigned by PIA, e.g. 43691) to a web server running on another box (10.1.2.10) and listening on port 8080.

I can access a simple page on my internal web server by accessing a url, e.g.:
http://myddnshost.myddnsprovider.net:43691

I added the following rules to my fw-script (the script generated by fw-config), right below the VPN forwarding rules. You can check out the full firewall script via this pastebin link:
https://pastebin.com/tRChJsH3

I created a separate rule chain for the port forwarding rule so that it can easily be flushed and updated whenever the external port changes (that would have to be managed by a custom script developed by the user) without clobbering other rules in the FORWARD chain. This new rule chain is linked to the VPN forwarding rule chain.
Since there is only one prerouting rule I put that in the PREROUTING chain, that too would need to be flushed and updated whenever the external port changes.

############### port forwarding rules

# prerouting rule:
iptables -A PREROUTING -t nat -i tun+ -p tcp --dport 43691 -j DNAT --to 10.1.2.10:8080

# create port forwarding rule chain:
iptables -N port_forwarding_vpn
iptables -t filter -A port_forwarding_vpn -i tun+ -p tcp -d 10.1.2.10 --dport 8080 -j ACCEPT

# add port forwarding rule chain to the vpn forwarding rule chain
iptables -A forward_rules_vpn -j port_forwarding_vpn

###############

Since there are too many variables involved in setting up port forwarding I'm not comfortable with adding logic to fw-config to interact with the user and configure port forwarding that way. I'd much rather just include a static example that gets added to the resulting fw-script (but is commented out) and add an 'Additional Information' page to the Wiki explaining the example.

mr-canoehead, 2018-05-06:
Private Internet Access has a simple API for requesting port forwarding, a call to the API turns on port forwarding and returns a JSON file containing your port number. They provide a script to do this, you can check out their script here:
https://privateinternetaccess.com/installer/port_forwarding.sh

Some scripting work would be required to automate the process of updating the firewall rules whenever the port changes, and for some applications it gets a little more complicated - e.g. transmission-daemon is configured with a peer-port that would need to be changed, and the transmission-daemon service needs to be restarted to pick up that change.

lehmanjo, 2018-05-06:
Thanks.  Have been browsing PureVPN's support documentation.  Haven't found much actionable content yet :)

mr-canoehead, 2018-05-15:
I have added examples for port forwarding using Private Internet Access, there are now two scripts included with the project for configuring port forwarding using a local static port (e.g. for an internal web server) and a dynamic port (transmission-daemon for seeding torrents). You can check out the new wiki page here:
[Port forwarding](https://github.com/mr-canoehead/vpn_client_gateway/wiki/Port-forwarding)

Let me know if we can close this issue on the basis of having added these example scripts and documentation to the project.

Thanks,
Chris

mr-canoehead, 2018-05-23:
Closing this issue based on having provided port forwarding examples + scripts for use with Private Internet Access.

closed 2018-05-23 by mr-canoehead