mr-canoehead/vpn_client_gateway #43: VPN Client Gateway is no longer compatible with Pi-hole

mr-canoehead, 2019-01-10:
nginx web server installation / configuration conflicts with the lighttpd service installed by Pi-hole. The VPN Client Gateway cannot be run on the same Raspberry Pi as Pi-hole. A revised configuration and corresponding installation instructions needs to be developed and tested.

farisoftfry, 2019-01-17:
Hello, I think there is a problem with dnsmasq too because Pi-hole don't use it anymore. It uses FTLDNS (that it is a sort of proprietary dnsmasq too) but dnsmasq is disabled by Pi-hole because both uses port 53 so the configuration to use the Pi as a Wifi Access Point doesn't work anymore. I advise to don't update to Pi-hole 4 until problems are solved. Cheers!

mr-canoehead, 2019-01-17:
I was able to resolve the issue with setting up the Pi as an access point, the problem was caused by the pihole-FTL service not being able to write DHCP lease info to /var/lib/misc. I resolved this by doing the following:
1) changed ownership of /var/lib/misc to root (it was statd:nogroup):
`sudo chown -R root:root /var/lib/misc`
2) installed the Access Control Lists package:
`sudo apt install -y acl`
3) granted read-write permission to the pihole user (pihole-FTL runs as 'pihole'):
`sudo setfacl -m 'u:pihole:rwx' /var/lib/misc`

Now the pihole-FTL service is able to write DHCP lease info to /var/lib/misc, and the service starts successfully.

I followed the Access Point instructions in the Wiki, but created a separate dnsmasq config file in /etc/dnsmasq.d named 'wifi_ap.conf' with the following contents:
```
interface=wlan0
no-dhcp-interface=eth0
dhcp-range=10.2.2.50,10.2.2.100,12h
```
I have managed to get Pi-hole and the VPN Client Gateway to run on the same Pi with nginx serving web pages for both systems without the need for lighttpd. When I have done a bit more testing with this setup I'll update the Pi-hole page in the Wiki with revised installation instructions.

So, my test Pi is currently doing the following:

 - running both Pi-hole and the VPN Client Gateway services
 - serving web pages for both systems via a single nginx site
 - acting as a Wifi access point, forwarding traffic via the VPN

mr-canoehead, 2019-01-20:
Updated the Pi-hole page in the Wiki with new instructions.
Updated the install guide with special instructions for Pi-hole installations.
Added nginx config file for combined Pi-hole + VPN Client Gateway systems.

closed 2019-01-20 by mr-canoehead