mr-canoehead/vpn_client_gateway #44: How connecting new device looks like?

patryk-smc, 2019-01-19:
I'm looking into getting started with this project. Does every client device need to be configured somehow before connecting or they will automatically connect via VPN automatically? Could someone elaborate?

mr-canoehead, 2019-01-20:
Closing this issue, the question was answered by Hobbabobba.
It would be helpful to create a basic network diagram to give new users an understanding of how the VPN Client Gateway will integrate into their network, I'll add this to my 'to-do' list.

closed 2019-01-20 by mr-canoehead

mr-canoehead, 2019-01-31:
I have created a network diagram for a basic installation of the VPN Client Gateway, you can view it via the following link:

https://gist.github.com/mr-canoehead/09d1dde3fd724f822edcb19cd95a6ed5

I would greatly appreciate any feedback you may have on it!

Thanks,
Chris