mr-canoehead/vpn_client_gateway #45: Failed to start VPN Client Gateway (vpncgw.service never wrote its PID file)

timcaffrey91, 2019-03-08:
I have issues getting the VPN Client Gateway to start. It gives me the following error after running command "sudo service vpncgw start":

`Job for vpncgw.service failed. See 'systemctl status vpncgw.service' and 'journalctl -xn' for details.
`

Output from the errors:

`-- Logs begin at Fri 2019-03-08 18:12:13 CET, end at Fri 2019-03-08 18:50:16 CET. --
Mar 08 18:50:07 raspberrypi sudo[3189]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Mar 08 18:50:08 raspberrypi systemd[1]: Starting VPN Client Gateway...
-- Subject: Unit vpncgw.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit vpncgw.service has begun starting up.
Mar 08 18:50:13 raspberrypi systemd[1]: PID file /run/vpncgw/vpncgw.pid not readable (yet?) after start-post.
Mar 08 18:50:13 raspberrypi systemd[1]: vpncgw.service: main process exited, code=exited, status=3/NOTIMPLEMENTED
Mar 08 18:50:13 raspberrypi systemd[1]: vpncgw.service never wrote its PID file. Failing.
Mar 08 18:50:13 raspberrypi systemd[1]: Failed to start VPN Client Gateway.
-- Subject: Unit vpncgw.service has failed
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit vpncgw.service has failed.
-- 
-- The result is failed.
Mar 08 18:50:13 raspberrypi systemd[1]: Unit vpncgw.service entered failed state.
Mar 08 18:50:13 raspberrypi sudo[3189]: pam_unix(sudo:session): session closed for user root
Mar 08 18:50:16 raspberrypi sudo[3255]: pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/journalctl -xn
Mar 08 18:50:16 raspberrypi sudo[3255]: pam_unix(sudo:session): session opened for user root by pi(uid=0)`

I installed pi-hole prior to installing vpncgw, though I uninstalled lighttpd to avoid any conflicts. 

Any tips?

mr-canoehead, 2019-03-08:
So the application service is failing to start. Nothing jumps out at me in the error log snippet, all I can suggest is that you carefully review section 3 of the installation guide (Install web server, application server, and web page files) and verify that all the required components have been installed / configured. The problem is not related to Pi-hole / lighttpd, I did a fresh install this morning by installing Pi-hole first (including the web server), then I uninstalled lighttpd and followed the installation guide. The application service started without any errors.

closed 2019-03-08 by mr-canoehead

timcaffrey91, 2019-03-10:
Thank you for the quick reaction. I have gone through the whole setup once more, reinstalling every package and triple checked if everything is installed/configured correctly. I keep getting the error. May need to start from a fresh installation. 

Once last thing - the "speedtest" package is not found/installed, giving me the only error in the setup process - could this have anything to do with it (and I should add a repo so I can have it installed)? Thanks again.

mr-canoehead, 2019-03-10:
Yes, the fact that the speedtest package is not installed will cause the vpncgw service to fail when it starts. Not sure why pip is not finding the speedtest package, but I've found that the speedtest-cli package also installs the required library. Try the following:
sudo pip install speedtest-cli

timcaffrey91, 2019-03-10:
That did the trick! Sorry to have put you through the trouble of mimicking the error. Thanks a lot for the efforts, I really appreciate it!!