mr-canoehead/vpn_client_gateway #46: Python 3.6

Mgrennes, 2019-05-10:
Hi,

Is this compatible with python 3.6? I'm getting some errors when starting the vpncgw.

`May  9 21:19:34 localhost env: File "/usr/local/lib/python3.6/site-packages/gunicorn/util.py", line 350, in import_app
May  9 21:19:34 localhost env: __import__(module)
May  9 21:19:34 localhost env: File "/opt/vpncgw/vpncgw.py", line 115
May  9 21:19:34 localhost env: except ValueError, e:
May  9 21:19:34 localhost env: ^
May  9 21:19:34 localhost env: SyntaxError: invalid syntax
May  9 21:19:34 localhost env: [2019-05-09 21:19:34 -0400] [14384] [INFO] Worker exiting (pid: 14384)
May  9 21:19:34 localhost env: [2019-05-09 21:19:34 -0400] [14378] [INFO] Shutting down: Master
May  9 21:19:34 localhost env: [2019-05-09 21:19:34 -0400] [14378] [INFO] Reason: Worker failed to boot.
May  9 21:19:34 localhost systemd: vpncgw.service: main process exited, code=exited, status=3/NOTIMPLEMENTED
`

Thanks! Great project also :)

mr-canoehead, 2019-05-14:
I have not tested the project with Python 3.6, all development work and testing is done with stock Raspbian which includes Python 2.7. When the Raspbian project updates to Python 3+ I'll run a full regression test and address any issues related to Python 3+ at that time.

closed 2019-05-14 by mr-canoehead

Mgrennes, 2019-05-14:
Hi, PIhole runs a debian based linux version. Since i'm installing it on a Centos 7 box, the script was looking for "service openvpn status" but it's actually "service openvpn@server status" for Centos 7.

Works great! Thanks!