mr-canoehead/vpn_client_gateway #47: 2 Clients connecting

Mgrennes, 2019-05-10:
Hi,

If BOX1 connects to the VPNCGW and opens a tunnel, can BOX2 connect to the VPNCGW and select a different TUNNEL?

Essentially, i'm looking to have 2 different boxes connected to the same VPNCGW, but with 2 different tunnels.

Thanks :)

mr-canoehead, 2019-05-14:
The VPN Client Gateway project only supports a single VPN tunnel. If you need multiple tunnels, a couple of approaches might be:
- set up multiple physical VPN Client Gateway servers, each running on its own Raspberry Pi, each connected to a different VPN tunnel
- set up multiple virtual VPN Client Gateway servers running in Debian virtual machines.
 
I have no plans to support multiple tunnels, as this would greatly increase the complexity of the project and would require significant work on all components (server app, web UI, iptables rules).

closed 2019-05-14 by mr-canoehead

Mgrennes, 2019-05-14:
Hi, Thanks for answering. This could be achieved by using ip rule tables. When a certain client connects identified by an IP, subnet range, tag (whatever), the script issues a ip rule to a certain table, identified by that IP, and creates a fowarding rule based on the IP. If this is something you would like to pursue, I can defintely help.