mr-canoehead/vpn_client_gateway #48: Can't run the vpncgw service

Mgrennes, 2019-05-15:
Hi,

I'm on python 2.7 and I can't get the program to run. Here's what i'm seeing:

`May 14 21:06:36 localhost env: [2019-05-14 21:06:36 +0000] [16318] [INFO] Starting gunicorn 19.9.0
May 14 21:06:36 localhost env: [2019-05-14 21:06:36 +0000] [16318] [INFO] Listening at: http://0.0.0.0:8000 (16318)
May 14 21:06:36 localhost env: [2019-05-14 21:06:36 +0000] [16318] [INFO] Using worker: eventlet
May 14 21:06:36 localhost env: [2019-05-14 21:06:36 +0000] [16328] [INFO] Booting worker with pid: 16328
May 14 21:06:36 localhost env: [2019-05-14 21:06:36 +0000] [16328] [ERROR] Exception in worker process
May 14 21:06:36 localhost env: Traceback (most recent call last):
May 14 21:06:36 localhost env: File "/usr/lib/python2.7/site-packages/gunicorn/arbiter.py", line 583, in spawn_worker
May 14 21:06:36 localhost env: worker.init_process()
May 14 21:06:36 localhost env: File "/usr/lib/python2.7/site-packages/gunicorn/workers/geventlet.py", line 102, in init_process
May 14 21:06:36 localhost env: super(EventletWorker, self).init_process()
May 14 21:06:36 localhost env: File "/usr/lib/python2.7/site-packages/gunicorn/workers/base.py", line 129, in init_process
May 14 21:06:36 localhost env: self.load_wsgi()
May 14 21:06:36 localhost env: File "/usr/lib/python2.7/site-packages/gunicorn/workers/base.py", line 138, in load_wsgi
May 14 21:06:36 localhost env: self.wsgi = self.app.wsgi()
May 14 21:06:36 localhost env: File "/usr/lib/python2.7/site-packages/gunicorn/app/base.py", line 67, in wsgi
May 14 21:06:36 localhost env: self.callable = self.load()
May 14 21:06:36 localhost env: File "/usr/lib/python2.7/site-packages/gunicorn/app/wsgiapp.py", line 52, in load
May 14 21:06:36 localhost env: return self.load_wsgiapp()
May 14 21:06:36 localhost env: File "/usr/lib/python2.7/site-packages/gunicorn/app/wsgiapp.py", line 41, in load_wsgiapp
May 14 21:06:36 localhost env: return util.import_app(self.app_uri)
May 14 21:06:36 localhost env: File "/usr/lib/python2.7/site-packages/gunicorn/util.py", line 350, in import_app
May 14 21:06:36 localhost env: __import__(module)
May 14 21:06:36 localhost env: File "/opt/vpncgw/vpncgw.py", line 1, in <module>
May 14 21:06:36 localhost env: from flask import Flask, request, jsonify
May 14 21:06:36 localhost env: File "/usr/lib/python2.7/site-packages/flask/__init__.py", line 21, in <module>
May 14 21:06:36 localhost env: from .app import Flask, Request, Response
May 14 21:06:36 localhost env: File "/usr/lib/python2.7/site-packages/flask/app.py", line 23, in <module>
May 14 21:06:36 localhost env: from werkzeug.routing import BuildError, Map, RequestRedirect, Rule
May 14 21:06:36 localhost env: SyntaxError: unqualified exec is not allowed in function '_compile_builder' it contains a nested function with free variables (routing.py, line 948)
May 14 21:06:36 localhost env: [2019-05-14 21:06:36 +0000] [16328] [INFO] Worker exiting (pid: 16328)
May 14 21:06:36 localhost env: [2019-05-14 21:06:36 +0000] [16318] [INFO] Shutting down: Master
May 14 21:06:36 localhost env: [2019-05-14 21:06:36 +0000] [16318] [INFO] Reason: Worker failed to boot.
May 14 21:06:36 localhost systemd: vpncgw.service: main process exited, code=exited, status=3/NOTIMPLEMENTED
May 14 21:06:39 localhost systemd: vpncgw.service: control process exited, code=exited status=7
`

Thanks for any pointers :)

mr-canoehead, 2019-05-15:
Are you installing on a Raspberry Pi?
Which operating system?

Mgrennes, 2019-05-15:
I'm running this on CentOS Linux release 7.6.1810 (Core)
Thanks!

mr-canoehead, 2019-05-15:
What is the host? Is it hardware, i.e. a Raspberry Pi, or are you running it in a virtual machine?

Mgrennes, 2019-05-15:
Hi,

I'm running it inside a vmware vm.

mr-canoehead, 2019-05-16:
I made a quick attempt to at least reproduce the errors you're seeing by installing CentOS 7 on a Raspberry Pi and going through the VPN Client Gateway installation procedure. Let's just say I didn't make it all that far, and I've run out of time to devote to this issue. At this point I have to say that I'm unable to provide any further assistance with issues related to running the VPN Client Gateway in a CentOS environment, and that you'll have to proceed with troubleshooting the issue on your own.

Best of luck with your efforts.

closed 2019-05-16 by mr-canoehead

Mgrennes, 2019-05-16:
Not a problem! I'll find what's wrong and will let you know. Thanks!