mr-canoehead/vpn_client_gateway #50: Service unavailable Attempting to contact server...

Sebaszz451, 2019-08-20:
Before i start i have to bring my respect for the amazing guide, awesome work, thank you for doing this mr-canoehead.

I followed the guide too the letter, and i cant get my head around the message Service unavailable Attempting to contact server on the VPN client gateway management page, i can see all the ticks green but cant acces anything cause the page is greyed out, also the current vpn server is not loading, is stayes on requesting data, i looked in the logs and see this popping by from time to time:

Aug 20 19:09:14 PI env[521]:     reraise(exc_type, exc_value, tb)
Aug 20 19:09:14 PI env[521]:   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1949, in full_dispatch_request
Aug 20 19:09:14 PI env[521]:     rv = self.dispatch_request()
Aug 20 19:09:14 PI env[521]:   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1935, in dispatch_request
Aug 20 19:09:14 PI env[521]:     return self.view_functions[rule.endpoint](**req.view_args)
Aug 20 19:09:14 PI env[521]:   File "/opt/vpncgw/vpncgw.py", line 545, in request_handler
Aug 20 19:09:14 PI env[521]:     return jsonify(function_map(request.args.get('request'))())
Aug 20 19:09:14 PI env[521]:   File "/opt/vpncgw/vpncgw.py", line 183, in get_current_server
Aug 20 19:09:14 PI env[521]:     countryname = server_details['countryname']
Aug 20 19:09:14 PI env[521]: KeyError: 'countryname'

My knowledge is very poor on this part, i have no idea what this means?
I use Purevpn and have a simmelair config as the server.conf as stated on page 2.
openvpn is connected and all is working great.

Could someone point me in the right direction where i need to look to fix this?

Cheers ;-)

mr-canoehead, 2019-08-20:
Can you please provide the following additional information, it will be helpful for troubleshooting your issue:

1) OS release info:
	`cat /etc/os-release`
	`uname -a`

2) contents of /etc/openvpn/server.conf

Thanks,
Chris

Sebaszz451, 2019-08-21:
Yes of course, thank you for the quick reply.

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Linux PI 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux

client
dev tun0
proto udp4
remote vleu-be1-ovpn-udp.pointtoserver.com 53
resolv-retry infinite
keepalive 10 60
nobind
persist-key
redirect-gateway
cipher AES-256-CBC
auth sha1
tls-client
remote-cert-tls server
auth-user-pass /etc/openvpn/auth.txt
auth-retry nointeract
pull-filter ignore "auth-token"
comp-lzo
verb 1
reneg-sec 0
ca /etc/openvpn/ca.crt
tls-auth /etc/openvpn/Wdc.key 1
auth-nocache
management 0.0.0.0 7505

mr-canoehead, 2019-08-21:
Did you replace the default server list (/opt/vpncgw/vpnservers.xml) file with the one for PureVPN (distributed with the project files, located in application/vpn_providers/purevpn)? If not, that could explain why the app is failing to look up the current server "vleu-be1-ovpn-udp.pointtoserver.com" and get its details (country etc.). Note that I created and tested that PureVPN server list some time ago, they may have changed some server names or added/removed servers since then. Unfortunately I'm not able to test it with their service since I don't currently have an account with PureVPN.

Sebaszz451, 2019-08-22:
Thanks for your reply, i did change it but ill have a look at the config and edit it, thanks for pointing me in the right direction! 
Cheers

Sebaszz451, 2019-08-22:
Thanks again, it worked like a charm, editing the xml, its all working!

closed 2019-08-22 by Sebaszz451