mr-canoehead/vpn_client_gateway #52: nordvpn update

rjl6789, 2019-10-26:
Hi,

Thanks for this excellent project.  I'm using nordvpn.  The nord scripts and schema needed some tinkering in order to work.

I'm opening this issue as per the contributing guidelines to avoid a surprise pull request.

Please see my commit: f9a24e968498b6395749accdeb8570d980609a0f which would form the basis of the pull request. I'm not precious - please feel free to make changes yourself as you see fit.

Cheers

mr-canoehead, 2019-10-26:
Excellent! Please go ahead and submit a pull request for your changes. If you could also please take a look at the NordVPN section of the [VPN Providers](https://github.com/mr-canoehead/vpn_client_gateway/wiki/VPN-Providers#nordvpn) page of the wiki in case some changes are needed for those instructions. I've added you as a collaborator so that you'll be able to  edit the wiki if needed.

Thanks for contributing to the project!

Cheers,
Chris

mr-canoehead, 2019-10-28:
Closing this issue; the proposed changes have been merged.

Thanks,
Chris