mr-canoehead/vpn_client_gateway #58: vpncgw_monitor.py spams Pi-Hole with DNS queries for www.example.com

halocaridina, 2020-05-25:
Title reflects the issue: vpncgw_monitor.py spams Pi-Hole with DNS queries for www.example.com at the rate of ~7 per 7-9 seconds. This makes the Dashboard cluttered. Given that filtering of the Dashboard is limited given that these are considered "legitimate" in the eyes of the developers (see numerous requests for this feature and resistance to it by searching for "pihole exclude particular domain from dashboard"), is there an option to do so when calling vpncgw_monitor.py?

Thank you in advance!!

halocaridina, 2020-05-26:
Update: modified BIG_LOOP_INTERVAL_MS=5000 to BIG_LOOP_INTERVAL_MS=60000 in vpncgw_monitor.py and restarting the service surprisingly made no difference in the rate of queries to www.example.com.

mr-canoehead, 2020-05-26:
The system performs DNS queries on a frequent basis in order to update the status info on the management web page; the system is functioning as intended.

The vpncgw_monitor.py script is executed by the statusUpdater function in vpncgw.py, the frequency can be controlled by adjusting the socketio.sleep parameter on line 166 of vpncgw.py, that parameter is 5 seconds by default, meaning that the vpncgw_monitor.py script will be run approximately once every 5 seconds.

The looping logic in vpncgw_monitor.py is not used, it is present due to legacy reasons; before the Flask Python app was developed it ran as a standalone process.

closed 2020-05-26 by mr-canoehead

mr-canoehead, 2020-05-27:
After reading the Pi-hole dashboard code a bit I was able to filter out `www.example.com` from the dashboard query logs via a crude hack, it involves modifying the `getAllQueries` request handler in `/var/www/html/admin/api_FTP.php`.

The relevant code starts at line 315 of the file.

I added an array `$ignoreList` containing a list of domains to filter out, then added a `continue` statement that is executed if the returned row includes a domain in `$ignoreList`. Here's the updated code, I have included the surrounding code for context (the lines I added are marked with `/ ***`):

```
                 $return = getResponseFTL();
                 $allQueries = array();
		 // *** list of domains to ignore ***
                 $ignoreList = array(
                         "www.example.com",
                         "some.site.com",
                         "another.site.com"
                 );
                 foreach($return as $line)
                 {
                         $tmp = explode(" ",$line);
                         // UTF-8 encode domain
                         $tmp[2] = utf8_encode($tmp[2]);
                         // *** skip this query log entry if the domain is in the ignore list ***
                         if (in_array($tmp[2],$ignoreList)) continue;
                         // UTF-8 encode client host name
                         $tmp[3] = utf8_encode($tmp[3]);;
                         array_push($allQueries,$tmp);
                 }

                 $result = array('data' => $allQueries);
```