mr-canoehead/vpn_client_gateway #60: Have to "sudo service nginx restart" on every reset

izo1, 2020-09-10:
Hi there this is the third time I am trying to install the server, everything works so far and it connects on reboot (soft or hard) but the web server needs to be restarted for the web portal to work.

Is this normal?

mr-canoehead, 2020-09-10:
No this is not normal behavior; the services (nginx, vpncgw, openvpn) should all start automatically on reboot/cold boot. I reinstalled the project on my test RPi this morning using the latest Raspberry Pi OS Lite (2020-08-20), and am not seeing this issue. All services start normally after a reboot/cold boot. I have not received any other reports of this issue.

What is displayed when you try to access the management web page via a browser? Does it just fail to connect?

Are you running Raspberry Pi OS Lite?

After a reboot / cold boot, what is the status of the nginx service? Run the following command:

`sudo systemctl status nginx`

The output on my system is as follows:
```
● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-09-10 12:40:49 PDT; 1h 32min ago
     Docs: man:nginx(8)
  Process: 629 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
  Process: 630 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
 Main PID: 631 (nginx)
    Tasks: 5 (limit: 2184)
   CGroup: /system.slice/nginx.service
           ├─631 nginx: master process /usr/sbin/nginx -g daemon on; master_process on;
           ├─632 nginx: worker process
           ├─633 nginx: worker process
           ├─634 nginx: worker process
           └─635 nginx: worker process
```

Do you see any nginx related errors in `/var/log/syslog`?

Thanks,
Chris

izo1, 2020-09-10:
Hi thanks for the response.

Running Raspberry Pi 3B+ with latest Raspian OS (full version).

Here's my output:

> nginx.service - A high performance web server and a reverse proxy server
>    Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: en
>    Active: active (running) since Thu 2020-09-10 16:11:18 PDT; 4min 36s ago
>      Docs: man:nginx(8)
>   Process: 540 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process o
>   Process: 541 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=
>  Main PID: 542 (nginx)
>     Tasks: 5 (limit: 2184)
>    CGroup: /system.slice/nginx.service
>            ├─542 nginx: master process /usr/sbin/nginx -g daemon on; master_proc
>            ├─543 nginx: worker process
>            ├─544 nginx: worker process
>            ├─545 nginx: worker process
>            └─546 nginx: worker process

Some error snippets I found in /var/log/syslog/nginx

> 192.168.1.193 - - [08/Sep/2020:17:58:56 -0700] "GET / HTTP/1.1" 200 2617 "-" "M$
> 192.168.1.193 - - [08/Sep/2020:17:58:56 -0700] "GET /styles/index.css HTTP/1.1"$
> 192.168.1.193 - - [08/Sep/2020:17:58:56 -0700] "GET /scripts/socket.io.min.js H$
> 192.168.1.193 - - [08/Sep/2020:17:58:56 -0700] "GET /scripts/main.js HTTP/1.1" $
> 192.168.1.193 - - [08/Sep/2020:17:58:56 -0700] "GET /images/openvpn_logo_powere$
> 192.168.1.193 - - [08/Sep/2020:17:58:56 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:58:56 -0700] "GET /styles/index.css?v=0.06694$
> 192.168.1.193 - - [08/Sep/2020:17:58:56 -0700] "GET /images/status_icons/loadin$
> 192.168.1.193 - - [08/Sep/2020:17:58:56 -0700] "GET /images/status_icons/loadin$
> 192.168.1.193 - - [08/Sep/2020:17:58:57 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:58:59 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:04 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:09 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:14 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:19 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:24 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:29 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:34 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:39 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:44 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:49 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:54 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:17:59:59 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:04 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:09 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:14 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:19 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:24 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:29 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:34 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:39 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:44 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:49 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:54 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:00:59 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:04 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:09 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:14 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:19 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:24 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:29 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:34 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:39 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:44 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:49 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:54 -0700] "GET /socket.io/?EIO=3&transport$
> 192.168.1.193 - - [08/Sep/2020:18:01:59 -0700] "GET /socket.io/?EIO=3&transport$
[... 173 lines cut ...]
> 2020/09/08 18:11:56 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:12:01 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:12:06 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:12:11 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:12:16 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:12:21 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:12:26 [error] 546#546: *2 connect() failed (111: Connection refus$
[... cut ...]
> 2020/09/08 18:13:26 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:13:31 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:13:36 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:13:41 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:13:46 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:13:51 [error] 546#546: *2 connect() failed (111: Connection refus$
> 2020/09/08 18:13:56 [error] 546#546: *2 connect() failed (111: Connection refus$

mr-canoehead, 2020-09-11:
I was able to reproduce this issue by installing the 'full' version of Raspberry Pi OS ("Raspberry Pi OS 32-bit with desktop and recommended software"). It is caused by the fact that in this version of Raspberry Pi OS the hostname 'localhost' initially resolves to the IPv6 address ::1 instead of the IPv4 address 127.0.0.1. Not sure why the resolved address switches to IPv4 after restarting nginx.

There are a number of different ways to resolve the issue, including:
1) installing the project using Raspberry Pi OS Lite (as specified in the Wiki)
2) disabling the IPv6 stack via /etc/sysctl.conf
3) changing the proxy address in /etc/nginx/sites-available/vpncgw.conf from 'localhost' to '127.0.0.1'
4) changing the bind directive in the gunicorn service config file from '0.0.0.0:8000' to '[::]:8000' (this causes gunicorn to listen on both the IPv6 and IPv4 local address)

Option 3 is probably the most straightforward. To apply this change, do the following:
edit the nginx site configuration file:

`sudo nano /etc/nginx/sites-available/vpncgw.conf`

change the `proxy_pass` setting to:

`proxy_pass              http://127.0.0.1:8000/socket.io/;`

After making this change, reboot the system and try accessing the management web page.

izo1, 2020-09-11:
Woow @mr-canoehead you're the best!
I will reinstall the whole with RPi OS Lite again in a little while, since it's working so perfectly right now.

Before I had issues with getting TorGuard to work so I had to sign up with PIA. 

Thanks much!

mr-canoehead, 2020-09-11:
I wouldn't call this a bug, I'd chalk it up to some sort of configuration difference between the two versions of Raspberry Pi OS. I always use the Lite version for my network appliance projects because it has a minimal amount of software included and it tends to have the fewest changes between releases.

It's possible to get this project working with TorGuard since they support OpenVPN, it would require development of a configuration file and server list XML file specific to TorGuard. PIA is the easier option since it is already supported by the project.

Chris

closed 2020-09-11 by mr-canoehead