mr-canoehead/vpn_client_gateway #62: can not start vpncgw service (Debian 10)

Hobbabobba, 2020-11-29:
I tried to install the vpn gateway on a new Debian 10 installation (vm).

When i try to start it with `sudo service vpncgw start` and get a status with `sudo service vpncgw status` i get

```
● vpncgw.service - VPN Client Gateway
   Loaded: loaded (/etc/systemd/system/vpncgw.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-11-29 10:23:57 CET; 1s ago
  Process: 9602 ExecStartPre=/bin/mkdir /run/vpncgw (code=exited, status=0/SUCCESS)
  Process: 9603 ExecStartPre=/bin/chown -R www-data:www-data /run/vpncgw (code=exited, status=0/SUCCESS)
  Process: 9604 ExecStart=/usr/bin/env gunicorn vpncgw:application --worker-class eventlet --timeout 30 -w 1 --threads 12 -b 0.0.0.0:8000 --pid /run/vpncgw/vpncgw.pid (code=exited, status=3)
  Process: 9611 ExecStopPost=/bin/rm -rf /run/vpncgw (code=exited, status=0/SUCCESS)
 Main PID: 9604 (code=exited, status=3)

Nov 29 10:23:57 vpngate2 env[9604]:   File "/usr/local/lib/python2.7/dist-packages/engineio/__init__.py", line 3, in <module>
Nov 29 10:23:57 vpngate2 env[9604]:     from .client import Client
Nov 29 10:23:57 vpngate2 env[9604]:   File "/usr/local/lib/python2.7/dist-packages/engineio/client.py", line 2, in <module>
Nov 29 10:23:57 vpngate2 env[9604]:     from json import JSONDecodeError
Nov 29 10:23:57 vpngate2 env[9604]: ImportError: cannot import name JSONDecodeError
Nov 29 10:23:57 vpngate2 env[9604]: [2020-11-29 10:23:57 +0000] [9609] [INFO] Worker exiting (pid: 9609)
Nov 29 10:23:57 vpngate2 env[9604]: [2020-11-29 10:23:57 +0000] [9604] [INFO] Shutting down: Master
Nov 29 10:23:57 vpngate2 env[9604]: [2020-11-29 10:23:57 +0000] [9604] [INFO] Reason: Worker failed to boot.
Nov 29 10:23:57 vpngate2 systemd[1]: vpncgw.service: Main process exited, code=exited, status=3/NOTIMPLEMENTED
Nov 29 10:23:57 vpngate2 systemd[1]: vpncgw.service: Failed with result 'exit-code'.
```

i double checked every installation step - you have some idea where i can find a solution for this problem?

mr-canoehead, 2020-11-29:
I have never tried running this project using Debian (only Raspberry Pi OS) so I spun up a Debian 10 VM today to try it out. When I tried to start the `vpncgw` service I encountered the same error as you did. I think it's a Python version issue, the long term solution is to migrate the project to Python 3 but in the meantime I was able to patch the system to get it working. Here is the workaround I used:

1. install the `simplejson` package: 

    `sudo pip install simplejson`

2. modify the `engine.io` client:

    `sudo nano /usr/local/lib/python2.7/dist-packages/engineio/client.py`

    change line 2 to:

    `from simplejson import JSONDecodeError`

3. start the `vpncgw` service:

    `sudo systemctl start vpncgw`

That should do the trick, After applying this patch the system seems to be working as expected.

\- Chris

closed 2020-11-29 by mr-canoehead

mr-canoehead, 2020-12-02:
Update: I have migrated the project code to Python 3 and have updated the install guide, patching is no longer required for this issue. I installed a combined VPN Client Gateway + Pi-hole system in a Debian 10 VM, everything seems to be working as expected.