mr-canoehead/vpn_client_gateway #66: Requesting data... issue

fateles, 2020-12-27:
Hello,
I'm installing a new version from scratch on a new SD Card to replace my older version. I tried 2 times, one with PiHole and other without it, but I always have the requesting data on the web interface.
I entered all the installation lines as per the step-by-step. On both I have this error on _syslog_:

`piavpn env[5415]: The client is using an unsupported version of the Socket.IO or Engine.IO protocols (further occurrences of this error will be logged with level INFO)`

I tried older versions manually and just crashed even more with _"not implemented"_ errors.
Tried to install some modules but in the end just ended being worse with more erros. After the install I don't have any erros I can understand related to vpncgw.

Thank you for the help and all the work!

mr-canoehead, 2020-12-28:
I re-installed my test system today and was able to replicate this issue. To resolve it I downloaded the latest socket.io client script and patched the system. After a reboot the management web page loaded properly.

To patch your existing installation:
```
wget https://cdn.socket.io/socket.io-3.0.1.min.js
sudo cp socket.io-3.0.1.min.js /var/www/html/scripts/socket.io.min.js
sudo reboot
```
I have now updated the socket.io client script included with the project files.

Thanks,
Chris

fateles, 2020-12-28:
Hi Chris,

Thank you for your quick answer! Tried again, with a clean SD card, and PiHole + VPN Gateway is working perfectly! Great work! Have a great week!

fateles, 2020-12-28:
Just to add something regarding the documentation:
The change server list the vpnservers.xml location is wrong. It should be /opt/vpncgw for this newer version.

Thank you again for everything.

closed 2021-01-06 by mr-canoehead