mr-canoehead/vpn_client_gateway #76: Stuck on Requesting Data (2022)

izo1, 2022-03-21:
Hi there love this VPN with PIA, I had used it for over a year and all of a sudden it got stuck on "Requesting data..."

I reinstalled it completely from scratch following the guide again and same thing.

Here's what I have right now in the log. I am  not sure what to do now, looking for another VPN but would love to stay with this one. Thank you!

Sun Mar 20 21:44:44 2022 WARNING: Using --management on a TCP port WITHOUT passwords is STRONGLY discouraged and considered insecure
Sun Mar 20 21:44:44 2022 OpenVPN 2.4.7 arm-unknown-linux-gnueabihf [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [PKCS11] [MH/PKTINFO] [AEAD] built on Apr 28 2021
Sun Mar 20 21:44:44 2022 library versions: OpenSSL 1.1.1d  10 Sep 2019, LZO 2.10
Sun Mar 20 21:44:44 2022 NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sun Mar 20 21:44:44 2022 TCP/UDP: Preserving recently used remote address: [AF_INET]102.129.145.147:1198
Sun Mar 20 21:44:44 2022 UDP link local: (not bound)
Sun Mar 20 21:44:44 2022 UDP link remote: [AF_INET]102.129.145.147:1198
Sun Mar 20 21:44:44 2022 WARNING: this configuration may cache passwords in memory -- use the auth-nocache option to prevent this
Sun Mar 20 21:44:45 2022 WARNING: 'link-mtu' is used inconsistently, local='link-mtu 1558', remote='link-mtu 1542'
Sun Mar 20 21:44:45 2022 WARNING: 'cipher' is present in local config but missing in remote config, local='cipher AES-128-CBC'
Sun Mar 20 21:44:45 2022 [losangeles414] Peer Connection Initiated with [AF_INET]102.129.145.147:1198
Sun Mar 20 21:44:46 2022 OpenVPN ROUTE6: OpenVPN needs a gateway parameter for a --route-ipv6 option and no default was specified by either --route-ipv6-gateway or --ifconfig-ipv6 options
Sun Mar 20 21:44:46 2022 OpenVPN ROUTE: failed to parse/resolve route for host/network: 2000::/3
Sun Mar 20 21:44:46 2022 TUN/TAP device tun0 opened
Sun Mar 20 21:44:46 2022 /sbin/ip link set dev tun0 up mtu 1500
Sun Mar 20 21:44:46 2022 /sbin/ip addr add dev tun0 10.18.112.46/24 broadcast 10.18.112.255
Sun Mar 20 21:44:46 2022 /etc/openvpn/client/change_dns.sh up tun0 1500 1553 10.18.112.46 255.255.255.0 init
Sun Mar 20 21:44:46 2022 WARNING: OpenVPN was configured to add an IPv6 route over tun0. However, no IPv6 has been configured for this interface, therefore the route installation may fail or may not work as expected.

and

pi@vpn:~ $ sudo tail -f /var/log/syslog
Mar 20 21:23:56 vpn systemd[635]: Reached target Timers.
Mar 20 21:23:56 vpn systemd[635]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
Mar 20 21:23:56 vpn systemd[635]: Reached target Paths.
Mar 20 21:23:56 vpn systemd[635]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Mar 20 21:23:56 vpn systemd[635]: Reached target Sockets.
Mar 20 21:23:56 vpn systemd[635]: Reached target Basic System.
Mar 20 21:23:56 vpn systemd[635]: Reached target Default.
Mar 20 21:23:56 vpn systemd[635]: Startup finished in 269ms.
Mar 20 21:23:56 vpn systemd[1]: Started User Manager for UID 1000.
Mar 20 21:23:56 vpn systemd[1]: Started Session 1 of user pi.
Mar 20 21:24:28 vpn systemd[1]: Started Session 3 of user pi.
Mar 20 21:25:49 vpn systemd[1]: Reloading.
Mar 20 21:25:49 vpn systemd[1]: getty@tty1.service: Current command vanished from the unit file, execution of the command list won't be resumed.
Mar 20 21:25:52 vpn systemd[1]: Reloading.
Mar 20 21:26:01 vpn systemd[1]: Starting VPN Client Gateway...
Mar 20 21:26:01 vpn systemd[1]: Started VPN Client Gateway.
Mar 20 21:26:02 vpn env[798]: Error: class uri 'eventlet' invalid or not found:
Mar 20 21:26:02 vpn env[798]: [Traceback (most recent call last):
Mar 20 21:26:02 vpn env[798]:   File "/usr/lib/python3/dist-packages/gunicorn/util.py", line 135, in load_class
Mar 20 21:26:02 vpn env[798]:     mod = import_module('.'.join(components))
Mar 20 21:26:02 vpn env[798]:   File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
Mar 20 21:26:02 vpn env[798]:     return _bootstrap._gcd_import(name[level:], package, level)
Mar 20 21:26:02 vpn env[798]:   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
Mar 20 21:26:02 vpn env[798]:   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
Mar 20 21:26:02 vpn env[798]:   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
Mar 20 21:26:02 vpn env[798]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
Mar 20 21:26:02 vpn env[798]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
Mar 20 21:26:02 vpn env[798]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
Mar 20 21:26:02 vpn env[798]:   File "/usr/lib/python3/dist-packages/gunicorn/workers/geventlet.py", line 23, in <module>
Mar 20 21:26:02 vpn env[798]:     from eventlet.wsgi import ALREADY_HANDLED as EVENTLET_ALREADY_HANDLED
Mar 20 21:26:02 vpn env[798]: ImportError: cannot import name 'ALREADY_HANDLED' from 'eventlet.wsgi' (/usr/local/lib/python3.7/dist-packages/eventlet/wsgi.py)
Mar 20 21:26:02 vpn env[798]: ]
Mar 20 21:26:03 vpn systemd[1]: vpncgw.service: Main process exited, code=exited, status=1/FAILURE
Mar 20 21:26:03 vpn systemd[1]: vpncgw.service: Failed with result 'exit-code'.
^X^C
pi@vpn:~ $ sudo tail -f /var/log/syslog
Mar 20 21:26:02 vpn env[798]:   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
Mar 20 21:26:02 vpn env[798]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
Mar 20 21:26:02 vpn env[798]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
Mar 20 21:26:02 vpn env[798]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
Mar 20 21:26:02 vpn env[798]:   File "/usr/lib/python3/dist-packages/gunicorn/workers/geventlet.py", line 23, in <module>
Mar 20 21:26:02 vpn env[798]:     from eventlet.wsgi import ALREADY_HANDLED as EVENTLET_ALREADY_HANDLED
Mar 20 21:26:02 vpn env[798]: ImportError: cannot import name 'ALREADY_HANDLED' from 'eventlet.wsgi' (/usr/local/lib/python3.7/dist-packages/eventlet/wsgi.py)
Mar 20 21:26:02 vpn env[798]: ]
Mar 20 21:26:03 vpn systemd[1]: vpncgw.service: Main process exited, code=exited, status=1/FAILURE
Mar 20 21:26:03 vpn systemd[1]: vpncgw.service: Failed with result 'exit-code'.
Mar 20 21:28:45 vpn systemd[1]: Unmounting RPC Pipe File System...
Mar 20 21:28:45 vpn systemd[1]: Stopped target Bluetooth.
Mar 20 21:28:45 vpn systemd[1]: Condition check resulted in Turns off Raspberry Pi display backlight on shutdown/reboot being skipped.
Mar 20 21:28:45 vpn systemd[1]: Stopped target Multi-User System.
Mar 20 21:28:45 vpn systemd[1]: openvpn.service: Succeeded.
Mar 20 21:28:45 vpn systemd[1]: Stopped OpenVPN service.
Connection to 192.168.1.50 closed by remote host.
Connection to 192.168.1.50 closed.
davidbadounts@Mac-Pro-2 ~ % ssh -l pi 192.168.1.50
pi@192.168.1.50's password:
Linux vpn 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Mar 20 21:39:22 2022 from 192.168.1.6

mr-canoehead, 2022-03-22:
I haven't used this project in some time, so I did a fresh install today to see if I could reproduce the problem you're encountering; yes, it's definitely broken. There seems to be an incompatibility between the latest versions of the Green Unicorn WSGI server and the Eventlet async handler.

I have encountered this issue before on another project and was able to resolve it by switching from 'eventlet' handlers to 'threading' in the gunicorn config + application server code. It's a minor change, so I modified my installation to get rid of the Eventlet dependency and now the web interface is working normally; I can see the vpn client gateway status, switch servers, etc.

I still need to do a bit more testing, and there may be some other issues to address as well (e.g. the 'Get IP address geolocation' tool seems to be broken). I hope to push the changes to a test branch later this week, and if all goes well merge it to the master branch this weekend.

I'll update this thread with my progress on the issue.

Thanks,
Chris

izo1, 2022-03-23:
Ah so sorry for pulling you back in haha!

If you have any alternate lightweight versions that are awesome as this one, possible to recommend one?

mr-canoehead, 2022-03-27:
An easy workaround for this issue is to install an older version of eventlet:
1) remove eventlet:
`sudo pip3 uninstall eventlet`

2) install eventlet 0.30.2:
`sudo pip3 install eventlet==0.30.2`

3) restart the application server:
`sudo systemctl restart vpncgw.service`

After doing this on my test system the management web page functioned as expected.

izo1, 2022-03-30:
> 

Thank you for that! I think it worked.

I believe these were auto updated when I had the server running for about a year?

Is there a way to disable auto update or is that off by default?

I guess I wasted time reinstalling everything haha

izo1, 2022-04-01:
@mr-canoehead I am also having an issue with DNS servers for PIA? It seems they keep changing it.

209.222.18.222 and 209.222.18.218 seems wrong and 1.1.1.1/8.8.8.8 may causee leaking.

Any other suggestions?

zweid0bby, 2023-06-26:
Any idea how to fix this or an alternative ??
sudo pip3 install eventlet==0.30.2 didnt work for me